topikachu/python-ev3 #43: I advance

vhf974, 2015-02-13:
Hello,

Thank you, yes indeed I understand a little better. I managed to run 'test_ev3_led.py'

But I have a mistake here:

root @ ev3dev: ~ # workon ev3_py27
(ev3_py27) ev3dev root @: ~ # python
Python 2.7.8 (default, October 18, 2014, 5:24:08 p.m.)
[GCC 4.9.1] is linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> From ev3.lego import MediumMotor
>>> D = MediumMotor ()
Traceback (most recent call last):
   File "<stdin>", line 1, in <module>
   file"/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/lego.py", line 172, in __init__
   file"/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", line 323, in __init__
ev3.ev3dev.NoSuchMotorError: No such motor type = port = minitacho
>>>

Thank you.

Good day.

topikachu, 2015-04-01:
well, seems something broken. Which ev3dev do you use?

topikachu, 2015-05-18:
I close this issue. Please reopen if the problem happens again.

closed 2015-05-18 by topikachu