topikachu/python-ev3 #44: Absolute IMU sensor: device or resource busy

jdchristensen, 2015-02-16:
I have a Mindsensor Absolute IMU gyro/tilt/compass sensor connected to my EV3 which is running ev3dev.  I can adjust the mode and read the values by directly accessing files in /sys.  But when I run the following program

    from ev3.mindsensors import AbsoluteIMU
    d = AbsoluteIMU(2)
    print(d.version)

I get the following backtrace

    Traceback (most recent call last):
      File "Abs_IMU_test.py", line 8, in <module>
        print(d.vendor_id)
      File "/usr/local/lib/python2.7/dist-packages/ev3/mindsensors.py", line 12, in vendor_id
        return self.read_byte_array_as_string(0x08, 8)
      File "/usr/local/lib/python2.7/dist-packages/ev3/ev3dev.py", line 418, in read_byte_array_as_string
        return ''.join(chr(r) for r in self.read_byte_array(reg, _len))
      File "/usr/local/lib/python2.7/dist-packages/ev3/ev3dev.py", line 415, in read_byte_array
        return [self.read_byte_data(reg + r) for r in range(_len)]
      File "/usr/local/lib/python2.7/dist-packages/ev3/ev3dev.py", line 389, in proxied_smb_method
        return getattr(self.b, method)(self.addr, *args, **kwargs)
    IOError: [Errno 16] Device or resource busy

The ev3dev people say that this is because python-ev3 is trying to use I2C commands to connect to the sensor instead of using the lego-sensor driver class.  It can't connect, since the ev3dev driver has taken control of the device.  Would changing this be difficult?

topikachu, 2015-04-01:
Which ev3dev version do you use?

jdchristensen, 2015-04-01:
It was downloaded and installed on December 13, 2014, so I believe it is ev3dev-jessie-2014-12-01.

topikachu, 2015-05-18:
Find the reason. The new ev3dev already recognize this sensor and has its native driver. I will update the code to use the new one.

topikachu, 2015-05-20:
fixed in the new 0.0.2.9 release

closed 2015-05-20 by topikachu