topikachu/python-ev3 #46: UltrasonicSensor measures not in CM but MM

HansAarneLiblik, 2015-05-03:
Measurements in UltrasonicSensor.dist_cm are not in centimeters but rather in millimeters. EV3 Ultrasonic Sensor max is supposed to be 255cm and dist_cm returns 2550

cavenel, 2015-05-03:
If you look at ev3dev documentation here
http://www.ev3dev.org/docs/sensors/lego-ev3-ultrasonic-sensor/ you can see
that it is in cm with one digit after floating point. Which means a value
of 2550 is 255.0 cm. I agree that the name dist_cm is confusing thought.

HansAarneLiblik, 2015-05-03:
Aha, cheers !

topikachu, 2015-05-12:
so not an issue? reopen again if you have any new idea. :)

topikachu, 2015-05-18:
I don't like the code now. I want to change the behavior to return the actual value. However, this may breaks existing behavior.

topikachu, 2015-05-20:
fixed in the new 0.0.2.9 release

closed 2015-05-20 by topikachu