topikachu/python-ev3 #47: Latest ev3dev release problems

xXx360noScopesMLGPROxXx, 2015-05-04:
Latest ev3dev release (2015-05-01), attempting to do example code yields:
```
root@ev3dev:~# workon ev3_py27
(ev3_py27)root@ev3dev:~# python
Python 2.7.9 (default, Mar  1 2015, 13:52:09) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from ev3.lego import MediumMotor
>>> d=MediumMotor()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/lego.py", line 172, in __init__
  File "/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", line 323, in __init__
ev3.ev3dev.NoSuchMotorError: No such motor port= type=minitacho
```
I'm not sure if that is intended behaviour or not because its mentioned in the README that python-ev3 is developed for 2014-12-01 release. Could you please clarify if there's a way to make it work in the latest release or if its required to downgrade the system in order to use it?

topikachu, 2015-05-12:
there's some api broken in the new image. will try to fix it.  Thanks.

topikachu, 2015-05-17:
a new release is published.

closed 2015-05-17 by topikachu