topikachu/python-ev3 #48: motors do not work on newest (2015-05-01) image

dekoza, 2015-05-11:
I tried to use python-ev3 on newest (2015-05-01) version of evdev. Sensors, sound, lcd and leds do work but motors fail to work. When I try to instantiate MediumMotor or LargeMotor I get NoSuchMotorError. Forcing port does not help. I am aware of issue #28

dekoza, 2015-05-11:
I checked the path in /sys/class:
# ls /sys/class/tacho-motor/
motor0  motor1  motor2

dekoza, 2015-05-11:
I started debugging motors by hand as described [here](http://www.ev3dev.org/docs/tutorials/tacho-motors/) and they work perfectly. I'm starting to debug/refactor the library.

dlech, 2015-05-11:
CC: @sharived (ref: ev3dev/ev3dev#323).

sharived, 2015-05-12:
I just put in a pull request for the changes I made to get the motors working.

topikachu, 2015-05-17:
@sharived @dlech 
Thanks a lot. A new release is published.

closed 2015-05-17 by topikachu