topikachu/python-ev3 #50: Sensor Problems

mipreuss, 2015-05-14:
i tried to work wit a sensor which can found is in the lego-sensor path. And get this error Message:

>>>ls =  LegoSensor(name='ms-light-array')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", line 229, in __init__
ValueError: invalid literal for int() with base 10: '2:i2c10'

with i2c i get the device is busy message because ev3dev has a driver for this Sensor

topikachu, 2015-05-17:
Is it possible to use the port number than the sensor name as test?

mipreuss, 2015-05-17:
i tried this too

>>> from ev3.ev3dev import LegoSensor
>>> l = LegoSensor(port=2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", line 233, in __init__
ev3.ev3dev.NoSuchSensorError: No such sensor port=2 name=None

>>> l = LegoSensor(port=2, name = 'ms-light-array')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", line 233, in __init__
ev3.ev3dev.NoSuchSensorError: No such sensor port=2 name=ms-light-array

the port_name for the Sensor on port 2 is:

root@ev3dev:/sys/class/lego-sensor# cat sensor0/port_name
in2:i2c10

on Port 1:
cat sensor2/port_name
in1:i2c10

Port 4: 
cat sensor3/port_name
in4:i2c10

topikachu, 2015-05-18:
Get it. 
The portname of iic sensor is different from others. :(
I have no such sensor. I can modify the source, but I can't test it.
Are you willing to test my new code when available?

topikachu, 2015-05-18:
The code is updated but not published.
Please
1. pull and checkout the master branch.
2. export PYTHONPATH=/path/to/python-ev3
3. run your code

mipreuss, 2015-05-18:
it works

>>> from ev3.ev3dev import LegoSensor
>>> l = LegoSensor(name='ms-light-array')
>>> l.value0
24
>>> l.value7
33

Thanks

closed 2015-05-18 by mipreuss