topikachu/python-ev3 #51: I can't controll the the motors in python

Oxel40, 2015-05-16:
Hi!
When i run this script in Python 2.7:
```
from ev3.ev3dev import Motor
d=Motor(port=Motor.PORT.A)
d.run_forever(speed_sp=50)
d.start()
```
No Errors appear but nothing is happening.
But it says that the motor is connected:
```
root@ev3dev:~# for f in /sys/class/tacho-motor/*; do echo -n "$f: "; cat $f/port_name; done
/sys/class/tacho-motor/motor0: outA
/sys/class/tacho-motor/motor1: outB
/sys/class/tacho-motor/motor2: outC
```
and it's working outside of Python:
```
root@ev3dev:~# export MC=/sys/class/tacho-motor/motor2
root@ev3dev:~# echo 50 > $MC/duty_cycle_sp
root@ev3dev:~# echo run-forever > $MC/command
```

I've just started programing on my ev3 and don't know why it's not working.

cavenel, 2015-05-16:
Hi,
By default, the `run_forever` mode will run with `speed_regulation = off`. If you look at the [tacho motors documentation](http://www.ev3dev.org/docs/drivers/tacho-motor-class/), it says that when `speed_reguulation` is `off`, it uses `duty_cycle_sp` to control speed. `speed_sp` is only used when `regulation_mode` is on.
On your first example in Python you use `speed_sp` and in your second example, outside Python, you use `duty_cycle_sp`. I think the difference is here!
You can use:
```Python
from ev3.ev3dev import Motor
d=Motor(port=Motor.PORT.A)
d.run_forever(duty_cycle_sp=500)
d.start()
```
or
```Python
from ev3.ev3dev import Motor
d=Motor(port=Motor.PORT.A)
d.run_forever(speed_sp=50, speed_regulation="on")
d.start()
```

dlech, 2015-05-16:
@cavenel, you have mixed up the values assigned your duty cycle and speed setpoints. It should be

    d.run_forever(duty_cycle_sp=50)

and 

    d.run_forever(speed_sp=500, speed_regulation="on")

@Oxel40: The value of the speed setpoint is different because the units are in "tacho counts per second" The NXT and EV3 Large motors have a maximum speed of about 900 tacho counts per second, so setting the speed setpoint to 500 should give approximately the same speed as setting the duty cycle to 50 percent.

Oxel40, 2015-05-16:
I tried to use your answer @cavenel @dlech but I just get this error:
```
d.run_forever (speed_sp = 500, speed_regulation = "on")
Tracking (most recent call last):
File "<stdin>", line 1, in <module>
Photo "/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", Line 344, in run_forever
Photo "/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", line 101, where FSet
Photo "/root/.virtualenvs/ev3_py27/local/lib/python2.7/site-packages/python_ev3-0.0.2.6-py2.7.egg/ev3/ev3dev.py", Line 185, in write_value
IOError: [Error Code 22] Invalid argument

```
and
```
d.run_forever (duty_cycle_sp = 50)
Tracking (most recent call last):
File "<stdin>", line 1, in <module>
TypeError: run_forever () takes exactly 2 arguments (1 given)

```
Why is this happening?

dlech, 2015-05-16:
Have a look at the [function definition](https://github.com/topikachu/python-ev3/blob/master/ev3/ev3dev.py#L334) of run_forever and you will see why.

Also which kernel are you running? You probably need these changes: <https://github.com/topikachu/python-ev3/pull/49>

Oxel40, 2015-05-16:
@dlech My kernel is: 3.16.7-ckt10-4-ev3dev-ev3

dlech, 2015-05-16:
Yes, there were some major changes to the motor drivers in that release, so you need the changes from the pull request that I mentioned to get python-ev3 working.

Oxel40, 2015-05-16:
Thanks alot @dlech!
But how do i change it/use the changes?

topikachu, 2015-05-17:
A new python-ev3 release is published with the latest change from @sharived .
Please reopne this issue if any new problem.

closed 2015-05-17 by topikachu