topikachu/python-ev3 #52: LEDs and some sensors not working in latest ev3dev

Potato42, 2015-05-23:
Hello!  I am using the latest version of ev3dev and am somewhat related problems with sensors and LEDs.

Sensors:
    Some of my sensors use different driver_names than the ones you look for.  In particular, I have no sensors that follow the pattern ev3-uart-[n]
    
    Of the sensors I have that are affected by this problem, these are the real names:
    Color sensor name is not ev3-uart-29, it is lego-ev3-color
    Infrared sensor name is not ev3-uart-33, it is lego-ev3-ir
    
    Looking at [http://www.ev3dev.org/docs/sensors/](http://www.ev3dev.org/docs/sensors/), the names that work for me are the ones currently listed, so it seems that the names were changed in a recent version of ev3dev (the one page archived on the wayback machine lists the old names)
    
    Until this is fixed, it can easily be prevented by using the port number of the sensor as the argument for the sensor constructor instead of -1.

LEDs:
    If you try to change the color of an LED, the following error occurs:
    ```
    >>> import ev3
    >>> led = ev3.ev3dev.LED()
    >>> led.right.color = led.COLOR.RED
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "/usr/local/lib/python3.4/dist-packages/ev3/ev3dev.py", line 481, in color
        assert 0 <= value[0] <= self.red.max_brightness
      File "/usr/local/lib/python3.4/dist-packages/ev3/ev3dev.py", line 98, in fget
        return property_type.post_read(self.read_value(name))
      File "/usr/local/lib/python3.4/dist-packages/ev3/ev3dev.py", line 56, in post_read
        return int(value)
    TypeError: int() argument must be a string or a number, not 'NoneType'
    ```
    This seems to be a result of the fact that /sys/class/leds/ev3:[color]:[side] is no longer being where led settings are stored, as they now follow the pattern /sys/class/leds/ev3-[side][0 if left 1 if right]:[color]:ev3dev as seen below:
    ```
    root@ev3dev:/sys/class/leds# ls
    ev3-left0:red:ev3dev  ev3-left1:green:ev3dev  ev3-right0:red:ev3dev  ev3-right1:green:ev3dev
    ```
    In an attempt to fix this, I subclassed LEDSide and changed the path it used, but this had no effect.  I don't really understand what's going on there, so I'll just leave you with the error.

topikachu, 2015-05-30:
which ev3dev  version do you use?

It seems the sensor name change again. :(

Potato42, 2015-05-30:
I'm using [http://www.ev3dev.org/news/2015/05/20/ev3-ev3dev-jessie-2015-05-20-release/](http://www.ev3dev.org/news/2015/05/20/ev3-ev3dev-jessie-2015-05-20-release/) (the most recent one), which does document all the changes I mentioned.

commit 77e731f5 (2015-06-06): Fix #52
New ev3dev change the uart driver name and LED file name.
closed 2015-06-06 by topikachu