topikachu/python-ev3 #53: Lego nxt light sensor

Oxel40, 2015-05-30:
Hi!
I wonder if there are any code examples for using the light sensor in batch or Python?
I can't understand how you would be able to take readings from the light sensor in bach or Python.
Thanks for all answers!!!

ddemidov, 2015-05-30:
Looks like you need [this class](https://github.com/topikachu/python-ev3/blob/master/ev3/lego.py#L16).

So the following should work:

```python
>> from ev3.lego import *
>>> s = LightSensor()
>>> s.reflect
>>> s.ambient
```

topikachu, 2015-05-30:
@Oxel40 
Does this class work for you? I close this issue now and feel free to reopen if any further problems

@ddemidov 
Thanks a lot.

Oxel40, 2015-05-30:
Thanks! The class works @topikachu.
But i have still a problem... should't the LED on the sensor be on when i use reflect?
And the sensor does only give the same value when i type s.reflect
```
>>> from ev3.lego import *
>>> s = LightSensor(port=3)
>>> s.reflect
115
```

Oxel40, 2015-05-30:
@topikachu I can't reopen this issue...

topikachu, 2015-06-06:
@Oxel40 Can you please check the mode and value0 at the same time?
eg,
cat /sys/class/lego-sensor/sensor0/mode
cat /sys/class/lego-sensor/sensor0/value0
However, you need to change 'sensor0' to your device number

Oxel40, 2015-06-06:
@topikachu here is the values:
```
root@ev3dev:~# cat /sys/class/lego-sensor/sensor1/mode
REFLECT
root@ev3dev:~# cat /sys/class/lego-sensor/sensor1/value0
1151
```

topikachu, 2015-06-06:
well, the python-ev3 is ok.
the mode and value are both correct.
will value0 change at some time?

Oxel40, 2015-06-06:
No, the value0 is always the same...

Oxel40, 2015-06-06:
@topikachu outside of ev3dev is the sensor working and the red LED starts. When i am using ev3dev the value is always the same and the LED won't start....

topikachu, 2015-06-08:
@Oxel40 Well, sorry, the LightSensor is http://shop.lego.com/en-US/Light-Sensor-9844. 9694 is a new one which is not supported yet . Please check http://www.ev3dev.org/docs/sensors/
I have no such sensor and I'm afraid you have to develop a new ev3dev sensor driver or wait for somebody to write one.
@dlech Do you have an idea of this sensor?

Oxel40, 2015-06-08:
@topikachu Thanks for all the help! 
Is there any instructions how to create a driver for a sensor?
This site http://www.ev3dev.org/contribute/#write-a-driver shows what you need but not how to write the driver it self.

dlech, 2015-06-08:
Sorry, we have never got around writing a driver for the NXT Color sensor. It is not an easy one to implement because it is unlike any other kind of sensor, so I keep putting it off.

Oxel40, 2015-06-08:
@dlech What makes the NXT Color sensor driver so different from the ev3 Color sensor driver?
If I want to try to write a driver for the NXT Color sensor, how do I write and use a driver?

dlech, 2015-06-08:
The EV3 color sensor is what we call a EV3/UART sensor meaning that it communicates using an EV3 specific protocol over [UART](https://en.wikipedia.org/wiki/Universal_asynchronous_receiver/transmitter). (See http://www.ev3dev.org/docs/sensors/ for a complete description of different types of sensors.)

All EV3/UART sensors use this same protocol, so once we implemented that, it is quite trivial to add individual sensor drivers. Same goes for NXT/I2C, NXT/Analog and EV3/Analog. The NXT Color sensor does not fit into any of these categories. It is kind of a hybrid analog and digital sensor that has it's own protocol. It also requires special treatment in the low level analog to digital converter driver, some of which is in place already.

So, to write a driver for it, you need to study the [LEGO firmware](https://github.com/mindboards/ev3sources) to see how LEGO did it and then translate that to fit the ev3dev driver framework.

If you would like to discuss this further, please open an issue at <https://github.com/ev3dev/ev3dev/issues>.

Oxel40, 2015-06-08:
Okay. Thanks for the help @dlech!
I'll see if I can manage to do something. I will let you know if I manage to make any progress with the driver.

topikachu, 2015-06-10:
I close this one. Please open a new issue if any other problem. Thanks.

closed 2015-06-10 by topikachu