topikachu/python-ev3 #55: Color Sensor (ev3-uart)

RideLikeTheWind, 2015-06-21:
I am trying to connect to the Lego Color Sensor. On the previous 2015-02-12 build, it came up red. Now, it doesn't come on at all. It is listed on the brick under ports as a eve-uart. When I try to connect, I get the following:

*I know there is a touch sensor on 3
>>> ls = LegoSensor(port=3,name='lego-ev3-color')
>>> ls.mode
'TOUCH'

*This is ok..

*Color sensor (supposed to be lego-ev3-color)
>>> ls = LegoSensor(port=1,name='lego-ev3-color')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build/bdist.linux-armv5tejl/egg/ev3/ev3dev.py", line 251, in __init__
ev3.ev3dev.NoSuchSensorError: No such sensor port=1 name=lego-ev3-color
>>> ls.mode
'TOUCH'

*Obviously not switching 

*No name, just port, same happens with just port - opens up -1
>>> ls = LegoSensor(port=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build/bdist.linux-armv5tejl/egg/ev3/ev3dev.py", line 251, in __init__
ev3.ev3dev.NoSuchSensorError: No such sensor port=1 name=None
>>> 

Any ideas?
I am running the latest EV3DEV build and latest Python-EV3

dlech, 2015-06-21:
Did you read the [release notes for the ev3-ev3dev-jessie-2015-05-20.img](https://github.com/ev3dev/ev3dev/blob/master/release-notes/ev3-ev3dev-jessie-2015-05-20.img-release-notes.md)? This is a known issue and the release notes explain how to fix it.

RideLikeTheWind, 2015-06-21:
Ah... I must have been more tired than I yhought. I'll check it out. My apologies.

topikachu, 2015-06-25:
Welcome. I close this issue now. If there's any further problem, please open a new one a drop me a mail.
Thanks

closed 2015-06-25 by topikachu