topikachu/python-ev3 #57: Reset Sensor value?

MaxNoe, 2015-07-11:
Is it possible to reset (or set) a sensor value?

I would like to null the gyro at program start and maybe several times in the program.

dlech, 2015-07-11:
It depends on the sensor. For example, the [LEGO EV3 Gyro sensor](http://www.ev3dev.org/docs/sensors/lego-ev3-gyro-sensor/) can be reset by momentarily changing to a different mode and then changing back to the desired mode. Most of the other gyro sensors out there don't have a reset function. You just have to maintain an offset variable in your program and use that to "zero" the sensor value.

MaxNoe, 2015-07-11:
Ah, ok thanks. I was talking about the LEGO Sensor.I thought i could avoid keeping track of the sensor value at certain positions.

closed 2015-08-03 by topikachu