topikachu/python-ev3 #58: why is the package namespace starting with ev3?

ensonic, 2015-07-16:
I have some add-on modules and I was structuring my packages like
```
project
  ev3
  nxt
  ...
```

if I now do
```
from project.ev3 import Xyz
```
and in project/ev3.py I have
```
from ev3.ev3dev import Lcd,Tone
```
I get import errors, as it seems that python now tries to resolve ev3.ev3dev relative to my project.

I'll probably just rename my stuff somehow, but I wonder if I am missing something here.

abkfenris, 2015-07-16:
If you are on Python 2, try
```python
from __future__ import absolute_import
```
Before any other imports, then it won't try to grab from within the package, unless you have `project.ev3` or `.ev3`.

ensonic, 2015-07-17:
@abkfenris Thanks! That makes it work. The wonders of python imports.

closed 2015-07-17 by ensonic