topikachu/python-ev3 #62: LED devices don't have delay_on/delay_off properties

ensonic, 2015-07-21:
The code in https://github.com/topikachu/python-ev3/blob/master/ev3/ev3dev.py#L464-465 exposes delay_on/off properties for the timing. Allthough:
```
ll /sys/class/leds/ev3-left0:red:ev3dev/
total 0
-rw-rw-r-- 1 root ev3dev 4096 Jul 21 15:15 brightness
lrwxrwxrwx 1 root root      0 Jan  1  2000 device -> ../../../leds_pwm
-r--r--r-- 1 root ev3dev 4096 Jan  1  2000 max_brightness
drwxr-xr-x 2 root root      0 Jan  1  2000 power
lrwxrwxrwx 1 root root      0 Jan  1  2000 subsystem -> ../../../../../class/leds
-rw-rw-r-- 1 root ev3dev 4096 Jul 21 15:15 trigger
-rw-rw-r-- 1 root ev3dev 4096 Jan  1  2000 uevent
```

dlech, 2015-07-21:
You have to set `trigger` to `timer` in order to create the `delay_on` and `delay_off` attributes in sysfs. This has some unfortunate side effects. See <https://github.com/ev3dev/ev3dev/issues/225> for more information.

My advice would be to implement your own blinking function rather than relying on the `timer` trigger.

topikachu, 2015-08-03:
@dlech Is https://github.com/ev3dev/ev3dev/wiki/Using-the-LEDs out of date? It still has delay_on/off

dlech, 2015-08-04:
The leds class is part of the mainline linux kernel, so it has not changed and is not likely to. You can use the `timer` trigger. I just don't recommend it because of permissions and race condition issues with the dynamic sysfs attributes (`delay_on`/`delay_off`).

dlech, 2015-08-04:
I suppose we should add a link to the issue I mentioned above to that page though.