topikachu/python-ev3 #63: Join forces?

antonvh, 2015-09-30:
I love python, and I love lego. I'm not so happy with the current boost-python implementation of ev3-dev. Neither is Ralph Hempel, he is working on a pure python branch here: https://github.com/ev3dev/ev3dev-lang/tree/pure-python. I want to contribute but I don't want to choose sides. Can't we all work on one python module? I think it's a matter of agreeing on namespace. If you are interested I can try to set up the repositories right.

MaxNoe, 2015-09-30:
I would highly welcome this, i was always confused which to use

bittner, 2016-01-01:
+1, though I find Ralph's repository very confusing. I'd prefer to have a clean repo for each language, one at a time. I can't see a compelling reason why to have a super-repo for all languages at all. And, of course, why are there submodules to various languages, but C++ is included as source code directly?

bittner, 2016-01-01:
A confusion example:

## Latest ev3dev installed

I've just installed `ev3dev` (`ev3-ev3dev-jessie-2015-12-30`) on a SDCard. I've managed to log in via SSH. Now the tedious part:

* Enter `dpkg -l | grep python` in the Terminal, and find `python-ev3dev` version 0.6.0 is installed. Check with Google: Good, this package is [on PyPI](https://pypi.python.org/pypi/python-ev3dev), nice! A link to the [package documentation](http://ddemidov.github.io/ev3dev-lang-python/) is also there, sweet. -- At first sight.

* Enter `python3` in the Terminal. Try first line of tutorial:
``` python
robot@ev3dev:~$ python3
>>> import ev3dev
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'ev3dev'
>>> 
```

* So, Python 3 is not supported? Try `python` in the Terminal instead:
``` python
robot@ev3dev:~$ python
>>> import ev3dev
>>> m = ev3dev.LargeMotor('outA')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'LargeMotor'
>>> dir(ev3dev)
['__builtins__', '__doc__', '__file__', '__name__', '__package__', '__path__']
```

WTF? There is a python package installed, but it doesn't do anything? And the documentation doesn't match.

* Check the documentation again. The [demos](https://github.com/ddemidov/ev3dev-lang-python/tree/master/demo) link leads to another repository. With more demos. Unfortunately, they don't work either. And - awesome! - in the README there is a deprecation warning: *"This version of python bindings has been deprecated in favor of rhempel/ev3dev-lang-python. This repository is no longer under active developement."* -- Not nice. And problem still unsolved.

* A [Web search](https://duckduckgo.com/?q=python-ev3dev&t=canonical&ia=about) for `python-ev3dev` yields *"topikachu / python -ev3 · GitHub"* as a second result, which looks very promising from the README. -- *"Maybe these are the current docs for the installed version 0.6.0?"*

... and so on.

I'd really see this repo to be the one included in the ev3dev image. It look much better maintained, much more in a good state.

G33kDude, 2016-01-01:
Running `help(ev3dev)` reveals that there are "Package Contents" `auto`, `brickpi`, `core`, and `ev3`. Importing `ev3dev.ev3`, as we're using the ev3, gets you the api you are looking for. Running `help(ev3dev.ev3)` reveals a bit more about the `ev3dev.ev3` module.

Following the documentation link on the pypi page you've provided leads me to [seemingly up to date documentation](https://python-ev3dev.readthedocs.org/en/latest/) with no link to the deprecated repository in sight.

Going through the list of unified libraries on [the official ev3dev website](http://www.ev3dev.org/docs/libraries/) brings me directly to the latest [official repository](https://github.com/rhempel/ev3dev-lang-python) by @rhempel, with a link to the same documentation.

bittner, 2016-01-01:
@G33kDude Okay, sorry. You are right. It's all in perfect shape. If you say so.

You could of course double-check to see if the points are valid. I'm not trying to be rude, just describing a *real* user journey. Try to double-check.

ddemidov, 2016-01-01:
@bittner reported the problem with documentation link at https://github.com/rhempel/ev3dev-lang-python/issues/99#issuecomment-168333231. I probably got it fixed before @GeekDude tried that, so both of you have valid points :).

G33kDude, 2016-01-01:
I never said it was in perfect shape. Also, you do have valid points. However, my point is that following the guides on the website brings you to where you need to look, and that we can't realistically tailor all the search engine results for a vague search like `python-ev3dev` (especially since the hyphen is likely ignored by most search engines).

rhempel, 2016-01-01:
Happy New Year everyone! Kreuzlingen - that sounded familiar! When I was a young boy my grandparents used to drive us up to the ferry crossing from Wangen Im Allgau to go shopping there.

So, I understand the frustration, and my goal for the weekend is to try and get my `ev3dev-lang-python` repository into good shape. So I will review Peter's comments carefully, try and add some more useful "getting started" docs and make the ecosystem safe for future contributions.

rhempel, 2016-01-01:
Peter, could I ask you to please copy your comments above (maybe trim after reviewing new information) and open a new issue at [https://github.com/rhempel/ev3dev-lang-python]. Thks.

pepijndevos, 2016-01-01:
Is any of the two libraries really much better than the other?

I think that the reason why it's one big repo is that they used to be SWIG bindings generated from one specification.

rhempel, 2016-01-01:
Well, that's all changed a bit in the past few weeks. We have split each binding into its own repo and made `ev3dev-lang` a submodule in each one.
At the end of the day, we are sticking with one "official" binding for each language.

bittner, 2016-01-01:
Don't get me wrong, everyone: The ev3dev project is awesome, really awesome. I apologize if my "user journey" reflected too much some obvious frustration. (I was trying to show my son the awesomeness of Linux and Python on his Lego brick. An attempt that failed in the end, despite his remarkable discipline to listen and wait.  :weary:)

@G33kDude The link to the libraries is indeed helpful though the page is a bit hard to find. Language bindings is an important topic for the Linux based Lego brick. It could be worth adding a direct link to the "Docs" menu of the website, what do you think?

rhempel, 2016-01-01:
HaHa! Having had three sons myself (all of which are now off to Uni) I can only give this advice:

1. Let your son play with the EV3 using the EV3-G GUI to get him interested.
2. When he goes to sleep, insert the ev3dev microSD card and play with Python on the EV3
3. When you are comfortable and have got his robot doing at least as well as he has using EV3-G - them show him Python

It's frustrating to watch someone muttering "This should work!" and "Argh - why are the docs so messed up" - better luck tomorrow - I'll be working on getting the basics cleaned up.

psychemedia, 2016-04-15:
One way of documenting the package could be to present working examples in a Jupyter (IPyhton) notebook. An IPython server can be run from the brick, although it's a bit slow, but I've also started exploring how we can run a remote IPython kernel on the brick and the heavier Jupyter server on a desktop/laptop machine on the same network as the brick. First doodles here: https://blog.ouseful.info/2016/04/15/thinking-around-the-edges-lego-ev3-robots-running-remote-jupyter-kernels/

I hope to produce some demo notebooks in the next couple of weeks documenting the library we have installed to support our own possible use of the library in a teaching context.

online4ever, 2016-04-17:
Hi everybody. I am using Windows 10, Python 3.5.1 and would like to develop some automation with a VisualStudio Python Project in order to get rid off boring things during System Test in my company. Unfortunately even I spent tons of time I could not solve the problem. I even registered to github to find an expert which could help me ;-) The bug reported month ago ist  still alive. What is a documentation worth if it does not work at all ;-(

>>> import ev3dev
>>> m = ev3dev.LargeMotor('outA')
Traceback (most recent call last):
  File "<pyshell#6>", line 1, in <module>
    m = ev3dev.LargeMotor('outA')
AttributeError: module 'ev3dev' has no attribute 'LargeMotor'

... does anybody have a good idea?

By the way ... Python supports EV3 installation with the following ...

pip install python-ev3dev

... entered from a command line.