topikachu/python-ev3 #65: Python 3: NameError: no such item __call__

ildan95, 2015-12-27:
When I try to init motor:
    d = LargeMotor()
it return error:
    NameError: no such item __call__

online4ever, 2016-04-17:
Hi everybody. I am using Windows 10, Python 3.5.1 and would like to develop some automation with a VisualStudio Python Project and Lego EV3 bot in order to get rid off boring things during System Test in my company. Unfortunately even I spent tons of time I could not solve the problem. I even registered to github to find an expert which could help me ;-)

. Python 3.5.1 (v3.5.1:37a07cee5969, Dec 6 2015, 01:38:48) [MSC v.1900 32 bit (Intel)] on win32
. Type "copyright", "credits" or "license()" for more information.
. >>> import ev3dev
. >>> m = ev3dev.LargeMotor('outA')
. Traceback (most recent call last):
. File "", line 1, in 
. m = ev3dev.LargeMotor('outA')
. AttributeError: module 'ev3dev' has no attribute 'LargeMotor'
. >>>

Does anybody know how to modify these code lines for working with my Lego EV3?

By the way ... Python supports EV3 installation with the following ...

pip install python-ev3dev

... entered from a command line.

liam-b, 2016-09-02:
Are you sure you've imported lang-python correctly? 
Also this should work:

```python
import ev3dev.ev3 as ev3
myMotor = ev3.LargeMotor('outA') # ports are from 'outA' to 'outD'
# do things with myMotor
```