topikachu/python-ev3 #69: Not able to work with motors

wilima, 2016-04-23:
Hi, testing on latest ev3dev and python 3, getting this error

`>>> from ev3.lego import *
>>> d = MediumMotor()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/robot/.virtualenvs/ev3_py34/lib/python3.4/site-packages/python_ev3-0.0.3.0-py3.4.egg/ev3/lego.py", line 217, in __init__
  File "/home/robot/.virtualenvs/ev3_py34/lib/python3.4/site-packages/python_ev3-0.0.3.0-py3.4.egg/ev3/ev3dev.py", line 330, in __init__
TypeError: 'NoneType' object is not subscriptable`

I am able to see motors in the folder /sys/class/tacho-motor/motor*/

ValorNaram, 2016-05-19:
run `cd /sys/class/tacho-motor` and `dir` and paste the output.
If you don't see motor0 and so on, then check the wires on the motor.
And try these steps:
1. shutdown your robot
2. disconnect all wires from your brick
3. boot ev3dev
4. if boot has finish, then connect the motors using the wires on your brick and run `cd /sys/class/tacho-motor` and `dir` again and paste the output too, if an output appears.

liam-b, 2016-09-02:
Are you sure you've downloaded and imported lang-python correctly? 
Also this should work:

```python
import ev3dev.ev3 as ev3
myMotor = ev3.LargeMotor('outA') # ports are from 'outA' to 'outD'
# do things with myMotor
```

dlech, 2016-09-03:
@liam-b, I think you have "python-ev3" (this repository) confused with "ev3dev-lang-python" (https://github.com/rhempel/ev3dev-lang-python)