topikachu/python-ev3 #73: Battery level in ev3dev

prabhu-deva, 2017-08-17:
Dear friends, 
    here I am doing a project on ev3 in which I had flashed ev3dev. Now I would like to  monitor the battery level of my brick any one can please help me to face this situation

BertLindeman, 2017-08-17:
I am not sure if this bash script is current due to the changing `/sys/class/leds/` file names.
But uoy could take a look at [checkpower](https://github.com/BertLindeman/bert-ev3dev-examples/blob/master/checkpower) it uses `blinkleds` that probably has older leds filenames. It depends on the kernel version. But that is easy to fix.

prabhu-deva, 2017-08-18:
Thank you,
    But my requirement is to know what is the current value of the battery level in the "**ev3 rechargeable battery**" which is come along with ev3kit.

BertLindeman, 2017-08-18:
I see that the script is no longer up to date and not yet compatible with the strech image I run.
But I *do* run an EV3 with rechargable battery and I do use an updated version of checkpower.
But the basic of it is, switch to the battery directory:
```
cd /sys/devices/platform/battery/power_supply/lego-ev3-battery
```
Maybe this exact directory name is not available on your level of ev3dev. 
You did not report at what level of ev3dev you run.
But it should be easy to find that out. E. g. using a find command like
```
find /sys/devices/platform/ -name voltage_now
```

In the `lego-ev3-battery` folder is a file `voltage_now` indicating the current delivered by the battery.

At the moment my EV3 reports with 
```
cat voltage_now
```
`8443733` which means 8.443733 Volt (I *think* it's in microVolt)

Would not be too dufficult for some savvy python programmer to use that info.

What made you conclude that checkpower script is not fit for the ev3 rechargable battery?

prabhu-deva, 2017-08-19:
I had gone through the root file but in first root I cant go further on battery at platform itself it shows an error that 
**robot@ev3dev:/sys/devices/platform$ cd battery /
-bash: cd: battery: No such file or directory**

BertLindeman, 2017-08-19:
As suggested above, you may try to find the `voltage_now` field using a `find` command like:
```
sudo find /sys/devices/platform/ -name voltage_now
```
At the moment I do nao have access to my EV3 brick due to bad sd-card.

prabhu-deva, 2017-09-04:
Thank you @BertLindeman  Here I can able to get the battery voltage at **/sys/devices/platform/legoev3-battery/power_supply/legoev3-battery$** and now I would like to display the value of the battery level from the above root folder.... could you please help me..

BertLindeman, 2017-09-04:
Sure ;-)

The fields as they are now on my ev3-brick running ev3dev at a stretch image:
```
ls -l /sys/devices/platform/battery/power_supply/lego-ev3-battery/
-r--r--r-- 1 root root 4096 Sep  4 11:35 current_now
lrwxrwxrwx 1 root root    0 Sep  4 11:35 device -> ../../../battery
drwxr-xr-x 2 root root    0 Sep  4 11:35 power
-r--r--r-- 1 root root 4096 Sep  4 11:35 scope
lrwxrwxrwx 1 root root    0 Sep  4 11:35 subsystem -> ../../../../../class/power_supply
-r--r--r-- 1 root root 4096 Sep  4 11:35 technology
-r--r--r-- 1 root root 4096 Sep  4 11:35 type
-rw-r--r-- 1 root root 4096 Sep  4 11:35 uevent
-r--r--r-- 1 root root 4096 Sep  4 11:35 voltage_max_design
-r--r--r-- 1 root root 4096 Sep  4 11:35 voltage_min_design
-r--r--r-- 1 root root 4096 Sep  4 11:35 voltage_now
```

The field you want is `voltage_now`.
```
# cat voltage_now
8474733
```
As you see you need to divide by a million. 
I use as an absolute minimum `6200000` so 6.2 Volt. 
If *that* level is reached the brick will poweroff quite quickly.

prabhu-deva, 2017-09-06:
yeah thanks here I can able to read the voltage by doing
   **cat voltage_now **     
     $ 7045800

but what I need is, the code to print the value of the battery voltage. which I get in those root folder to screen. but I cant able to get the value from the root folder.

BertLindeman, 2017-09-06:
Sorry, I do not use the EV3 screen, too smal for me.
So I can not help you with printing a value to the screen.

Maybe the examples on [this web page](https://sites.google.com/site/ev3python/learn_ev3_python/screen) can help
You may look at the example "Display Text" on that page.

prabhu-deva, 2017-09-07:
yeah, But my exact requirement is **the python code to get the value from the root folder** continuously 
**how to access the value of  root folder**